Anthilla/AnthC #3: 1st contribution result from @clearemc

dorvan, 2021-06-07:
Hello @dorvan , @Doenec ,

I am @clearemc we discussed about doing a first revision oriented to Electromagnetic Compatibility (EMC). 
First of all, congratulations for your work and for your efforts to make it an open hardware project! 
Thank you for letting me participate in it. 
I have done a general review of the schematic and indicated some modifications that can reduce problems during the certification phases. 
I have also indicated some points that an affect to the functionality of the board. 
Here it goes!

    Schematic
        Connector J3: SPI
            The 3 critical signals for SPI (MOSI, MISO, SCK) arrive all together to the connector. If possible, add a grounding pin in the middle of them, so you reduce crosstalk as much as possible. The ideal solution would be MOSI – GND – MISO – SCK – GND, but that would be too much space, so, try to isolate at least the SCK signal, because it will be always active during the communication.
        U7: ADDR pin à Datasheet states that it can be left floating, but since it is an input, it is better to connect it to ground to avoid susceptibility issues. This is applicable to all input pins that are floating, is it recommendable to tie them to ground or to a fixed voltage.
        U8:
            I7 input: if not used, better connect it to ground.
        J1: it is better to connect the shield to ground. I recommend you to leave the footprint for 1 or 2 components in parallel to ground, so you can add one or two capacitors, or even a RC circuit.
        U5:
            Pin 6 goes to a decoupling capacitor, but it is not connected to any power net. 
        Connector J4: 12 V line. Add some protecting components, such as a diode for polarity reversion protection and a TVS to limit transients and overvoltages.
        RTC:
            Add reverse polarity and overvoltage protection to the coin cell battery.
            The oscillator Y2 has not capacitors, it needs 12.5 pF capacitors to oscillate. 
        Power connector J5: if this is the general power input of the board, it will need protections for reverse polarity and also a common-mode and differential-mode filter.
        General:
            Communication and digital lines: it is nice to add a series resistor of about 100 ohms to control the slew rate of the communication lines, as well as to limit the current for the digital ones. It can be quite useful in tests such as susceptibility, ESD, or even to avoid crosstalk.
            Net I34 is not connected to resistor R52

Please, let me know if you have any question or remark. 
When possible, 
I will perform a review of the layout!

Kind regards
@clearemc

M10CUBE, 2021-06-14:
Hi guys. Continuing my conversation from Linkdinn with @dorvan.
Here is a summary.
We are building a community project (more an idea and hardware framework) with code name M10CUBE
Project leaves here:
M10CUBE project: https://hackaday.io/project/171770-m10cube
GitLAB : https://gitlab.com/m10cube/m10
The idea of the project is exactly as yours but different form factor. As you see on wiki we concluded on that form factor as a golden rule of size, and real estate on board to include I/O industrial interfaces and the same time to be modular based on Raspberry Pi 40pin bus 
Now I am designing the new version (new board size is 90x90mm and some small movement to mounting holes)
I am willing to contribute to your project transferring my experience of what I did hoping to make both projects better.
Your project is at the correct dimension and I am sure we share the same idea of what a universal controller must be (I have it all explained in my wiki page on git lab  as I said) . https://gitlab.com/m10cube/m10/-/wikis/home
For the shake of conversation let assume that everything I wanted to include on my M10CUBE form factor can be done on the RASPI form factor. I really like to see that. So first of all
- You need to have I/O isolation on your board as you see the three I/O boards on our project.
https://gitlab.com/m10cube/m10/-/tree/master/ELECTRICAL/INPUT%20OUTPUT
https://gitlab.com/m10cube/m10/-/tree/master/ELECTRICAL/OUTPUT
As I said to @dorvan there are two interesting chips from ST that meant for that purpose.
That is VNI8200XP and CLT01-38SQ7 chips and ST already put them on an Arduino (nucleo) PCB for promoting them
I have sent to  @dorvan on Linkedin the schematics of this board. The only drawback is that it is difficult to hand solder. That is why we did not use it on our M10CUBE design. But on your form factor you have limited space and when you want to design industrial I/O on RASPI that is the chip. Made in Europe.
That is for now and I relay want to discus how you have this project in your mind after one year. Can this support all our ideas (within limits of a controller of course. I am willing to participate and help without taking out of your road but I have many years of industrial automation experience and I really want to built something innovating. Not a streamline product

M10CUBE, 2021-06-15:
Github Issues (I use GitLAB) is not a great place to do teamwork.
The Best (and free) is Slack. We are using it to deploy the M10CUBE project.
Hackaday.io is another nice place to publish your work but I just know the basics
As I said 10BASE-T1L Ethernet is extraordinary. A few people know about it but it will revolutionize how IOT is interconnecting. It has everything we can desire.
I was surprised why they did not think that before.
I strongly recommend you to attend the free webinar.

Enabling Seamless Connectivity with 10BASE-T1L Ethernet

LIVE WEBCAST DATE: June 30, 2021
LIVE WEBCAST TIME: 03:00 PM CEST
Use the link below to enter the webcast up to 15 minutes before the start.
WEBCAST LINK: https://event.on24.com/wcc/r/3148687/2D3840291798029D0A659DD1DE46862D

USB type C is the future
Consider POE as well but 10BASE-T1L Ethernet transfers power via the 2 wires. And all that having seamless IP connectivity (business as usual) and NO USE of switch in every node. Amazing! I do not know how they did it
I am sure in near future all MPU will be 10BASE-T1L Ethernet ready as it is for CANBUS now.
I am sure Raspi Pico will be one of them.
I am also sure that I/O extenders (like the I2C we have now) will use this 10BASE-T1L Ethernet on chip

Regards

M10CUBE, 2021-06-15:
This article is very important to read if you have some time.  For IOT designers and visioners.
What is the AI of things (AIoT)?
https://www.embedded.com/what-is-the-ai-of-things-aiot/?utm_source=newsletter&utm_campaign=link&utm_medium=EmbeddedEurope-20210615

dorvan, 2021-06-17:
I confirm for: 

Try to condensing all the components possible to make more free space respecting all the point specified by:
- Current Design
- with addition of @clearemc additions added on main issue
- use of VNI8200XP and CLT01-38SQ7 chips **to enhance electrical specifications, simplyfing the layout and reducing component, only if permit to maintain the same form factor**
- considering 10BASE-T1L with POE  (**about POE to enhance electrical specifications, simplyfing the layout and reducing component, only if permit to maintain the same form factor**)
- USB type C passage in order **to enhance electrical specifications, simplyfing the layout and reducing component, only if permit to maintain the same form factor** Looking there: https://www.linkedin.com/feed/update/urn:li:activity:6810181646369878016/?commentUrn=urn%3Ali%3Acomment%3A(activity%3A6810181646369878016%2C6810495075337039872)
https://pallavaggarwal.in/learn-embedded-system/

Open questions about "General" Point in the issue:

Do you recommend to place one ground routing next to each digital signal?

- ask to a PCB company with Design Department a very resistant to high temp and difficult environment material to enhance PCB specs, maintaining compatitive costs.

- Make an Open Request to N PCB company for a competitive OPEN OFFER about costs to make design compliant to
    RoHS
    CE/FCC and most diffuse certifications
    common and specific EMI
    General/National EMS
    EN 61131-2:2007
    EN 61010-1:2010+A1:2019
    EN IEC 61010-2-201:2018
    with addition suggested.

M10CUBE, 2021-06-17:
From my point of I will tady your work done so far. I agrre with @dorvan;s confirmations and at the end of the month I will watch the webinar about the 10BASE-T1L. I am not sure but the specs say that power can be transmited by the same 2 data wires.
I will have more on that after the webinar.
Belive me when the project starts steaming a workplace to communicate will be necesarry.
Hackaday is one (and we can start with that) but by far as good as Slack which can be a complementary in a future rime)
My decision is to join full time with your project trying to see what can be done on the RASPI factor. The same time my M10CUBE form factor can borrow all or some of the technology learned here and applied to M10CUBE in due time. That way we can develop both platforms and we can learn a lot from the journey.  Mistakes can be done and corrected. At the end everybody will benefit from the experience.
For M10CUBE I choose CERN OSHW license. Verification code  [GR000004]([url](https://certification.oshwa.org/gr000004.html)). It is the best choice.

clearemc, 2021-06-17:
Hello,

@dorvan what are you looking for regarding the difficult environmental conditions? In terms of PCB material, you would need pre-pregs with high Tg and high CTI. Extreme conditions of temperature, humidity and, dust would force you to respect safety distances a little bit higher than in regular conditions and to pass dielectric tests with a higher restrictive humidity conditioning. 

From my point of view, it is early to ask for quotations about EMC/safety compliant designs, I recommend you first to close specifications, and perform a first design and testing. When preparing a certification you will need to provide all the information about how many inputs, outputs, type of signals, kind of cablings, describe a normal use of the system... many things that right now they are not defined.

M10CUBE, 2021-06-17:
I agree with @clearmc. Since we are on the developing stage it is waste of time to go for EMC/safety rules at this point. I thing we must make a ROAD MAP of the design. 
- We have something already build and we are going to change
- We must write down all the specs of that board
- To me looks just a HAT but do we want this to be part of a bigger project (modular system?). What are the plans?
- What is the final market for that?
- Above all detailed specs and of course to see if all these ideas have enoght real estate on the RASPI form factor. We adjust as we move ...
- ALL written down as accurate can be otherwise we will go back and forth.
- It is good to spend more time on planning and less time on the drawing board.

dorvan, 2021-06-18:
Hello @clearemc @M10CUBE the board it's son long and difficult development process we started from some development board years ago.  A lot of answers to questions are already defined in readme file of the project.

I'm discussing about features I've already requested to @Doenec in feature requirements. 

I can recap some details for you here

- Initially "logic section" and "power section" like oldworld macs will be separed on 2 boards to have more space and maintaining the RASP B+ form factor (that it's already a compromise because I want a smaller one factor, so bigger == no)
- yes, will be really operatively rock solid. And operative in high temp ranges. Was a first request, so yes the additional distances will be consodered or places cuts into pcb to make phisical separation
- optocouplers are present in the feature requirements. Compromises depends on space/time needed to go forward.
-markets/usage: industry, machine automation, home automation, public devices first.

I've wrote about a design update firstly considering:
- @clearemc changes implementation in design update in order to fill main projects needs: goes to production and be usable as product. 
- tune-up the design on the mainboard making more free space using some integrated to save space. (Same design specs)
- evaluate additions proposed by @M10CUBE if are good or necessary on the mainboard, or will be an Hat or Shield. If can be an additional board, will be a passthrough board.
- evaluate the possibility to change from USB micro to USB-C if this can tune-up and give + to the power section and permit to free some space. 

Priority is:
- tune-up/fix current design with all necessary things for the mainboard due go to production and permit usage as product.
- think about next release design and/or additional board.

What need to be absolutely present on mainboard?
- all you need to use this board in automation, in the 80% of cases.

This is not an university or a pure R&D project, but it's a project that want to go to production fastly and work on releases upgrade.

Thanks

M10CUBE, 2021-06-18:
I see your point @dorvant and we must follow your plans. Overall design is looking good and processional. I like it
Looking at the design (I may look harder or I mished something)
1-  I see one pin on 40pin RASPI BUS connector that is different then the RASPI pinout. That is pin 7 you have GND (maybe intentionally?) Original is  Raspberry GPIO4.
2 - I do not know if you inted to use it as a HAT (my opinion is you do that ESP32 HAT). In that case you must folow HAT specs. That is use the I2C flash memory on pins 27 and 28 so RASPI knows what is plugged on its bus. KICAD has a template for that
3 - USB C is a muc h better idea . That will be the new standard but IF you run out of space you can omit the USB programmer all together and use only 4 pins to program the ESP32 with an external programmer. If you still can find space type C must be there
4 - I am not really follow your consept with Inpust and Outputs. Why the out put is going on the RASPI 40pin connector?. That is sto drive a relay so it need to go to external connector not on the RASPI BUS. Same the Input consept. It confuses me. I am coming fro industrial automation design and one thing is clear. There is the "outside" normally 24V and the "inside" normally 5 or 3.7 V. RASPI BUS is considering "inside" world. That is why we are using interface chips.
Again I apologize but I must understand what exactly we mean by digital inputs and outputs to the "outside"

dorvan, 2021-06-19:
Thanks @M10CUBE for youn point of you, I'm trying to explain. 

About the GPIO will be only inspirated to RASP GPIO but obviously the platform spec are different so apart errors or thing can be cahnged to be more near to the standard, the GPIO it's only inspirated, but different like the spec.

The GPIO and the other connectors will the highway to extend platform spec and roles. Connectors to interconnect directly to outside world, GPIO it's to permit ot ESP32 HAT to extend platform with additional board to extend functioanlities but also to connect external dougher boards like sensors. 

About the voltage availability and grounding it's the same. I will be able to use 3.3/5v or 12-24v senction as role, and separately and without depending (as ppssibile) one from other. 

For example I wante to make a "smart button" it's enough the 5v and 3.3v section. This also to programnand test. The same for other implementation. Like a usage with relay. 

I considered in to have separate circuits external and internals to interchange with. The AnthC controller doesn't be a part of the resultant project circuit! This is primary rule and it's very important.
It's not the implementation of the project. It's the driver. It's the controller of what is implemented... 

Thinking about relays:
Example, you have to make a light automation. You set the circuit with it's power source, lights, and all the need.. then you use a din relay component with clean contacts and test it. All works great.
Ok now you take an AnthC board, power it in the best way for main powersource you want to use and in relation of the I/O you will use to drive relay with clean contacts. And all it's made and automated. Are many (separed) circuits that interact.

M10CUBE, 2021-06-19:
@dorvan I lost you from one point. You probably thinking something grate that iI can not digest at this moment. I like proposals that looks to me astrophysics at first look but I am sure you are talking something nobody thought before. OK lets make a reset and explain me one at a time please. You may loose some time but that will make me help you better. 
That will make me to fully understand your concept and to start helping the team
- 1 Do you want this board to considered as a HAT?. Thus following all HAT specs? (that is at least 40 pin connector must follow exactly RASPI bus, I2C memory on board to tell RASPI what is connected

dorvan, 2021-06-20:
@M10CUBE  not at all, I consider AnthC as independent platform, the baseline board it's the motherboard. Can have his HAT boards, as expansion boards, compatible partially or fully with it's GPIO format.

hwidvorakinfo, 2021-06-21:
> * [ ]  **U5**:
> * [ ]  **Pin 6 goes to a decoupling capacitor**, but it is not connected to any power net.

This pin 6 on CP2102 (and similar chips) is the output of the internal LDO. There is no need to change it. The CP2102 is being powered from the VBUS line.

hwidvorakinfo, 2021-06-21:
> * [ ]  **RTC**:
> * [x]  Add reverse polarity and overvoltage protection to the coin cell battery.
> * [x]  The oscillator **Y2** has not capacitors, _it needs 12.5 pF capacitors to oscillate_.

No, the caps are built-in in the RTC device.

How can be the coil cell flipped over in the socket? Is it really necessary?

clearemc, 2021-06-21:
Hello @hwidvorakinfo, it is a pleasure to work with you. Thank you for your comments, you are completely right regarding the CP2102 and the RTC.

Regarding the coin cell: the coin cell can be mounted in the wrong sense in the assembly line.  I know that it can sound useless to add protection components, but for safety certifications such as the 62368 or the 61010 (needed for the LVD), you are asked to show that every source of energy, even the smallest ones, do not provoke any hazard to the users or to the system. Thus, you need to perform tests or to show through the schematics that the system is still not dangerous when you short-circuit the coin-cell or when you install it in the wrong sense.

M10CUBE, 2021-06-21:
> @M10CUBE not at all, I consider AnthC as independent platform, the baseline board it's the motherboard. Can have his HAT boards, as expansion boards, compatible partially or fully with it's GPIO format.
Trying to understand and put my comments. Let me see what it is clear
- AntC DOES NOT follow RASPI HAT specs. That is because you have different format on the 40 pin connector and you have not I2C memory on board. That mean you **CAN NOT** plug a Rasp pi in this bus. This is true even if you use some of the pins same as RASPI
- That being said we must NOT call this board a HAT board
- AnthC is a motherboard with RASPI PCB dimension with a proprietary 40pin header. That is a motherboard and you want to build other boards on top of that. I suggest that this 40 pin is independent in every board and CAN NOT connect boards together using this 40 pins as a BUS . Only way to connect these boards is externally  using the RS485 connector. Can not use the 40 pin connector as a BUS. It can be user only as a header to get the digital or analog pins and drive other PCBs 

My comments are:
- First It is difficult to create a new standard. Extremely difficult other people to get it and use it with other popular platforms (raspberry). So that will have limited market value ans less people to contribute. If that is so I have my comments on the way pins on the 40 pin connector used.
- AnthC has interesting hardware on board but the overall hardware design does not follow hoe a home or industrial  or home modular controller must be.
- To give an explanation. A modular design must not have the "outside" external analog inputs on the back-plane bust.

I rely do not want to change your mind and I will follow your project until I see your final idea because I still can not follow your mind. That is because my experience and my education in building and using controllers more that 35 years tell me that something is wrong here.
Do not get me wrong. I will try to help you here. Market is changing fast. May be some time in future proved that you were correct and I was wrong. It is OK with me. 
In any case I will watch you affords and see how it goes. I believe that I will learn a lot. Things that can embed on M10CUBE and available to everybody

dorvan, 2021-06-26:
@M10CUBE  _"AntC DOES NOT follow RASPI HAT specs. That is because you have different format on the 40 pin connector and you have not I2C memory on board. That mean you CAN NOT plug a Rasp pi in this bus. This is true even if you use some of the pins same as RASPI"_  I know this perfectly, And I've wrote as answer to your questions.

M10CUBE, 2021-06-26:
Ok @dorvan . I have now the info needed. I better stay aside and watch instead of interfering with your design. A lot to learn specially because it is KiCAD designed and I have interest on that.
I will make suggestion only if I see an electrical or mechanical problem. I have a lot of work to do in M10CUBE as well.
I have to finish the new sensor board with mikrobus on board

clearemc, 2021-06-27:
Maybe there is a misunderstanding: HAT stands for Hardware Attached on Top. They are quite popular in the Raspberry Pi universe, but it does not mean that these kinds of structures cannot be used for different types of boards. Hence, the AntC board does not need to follow the Raspberry PI specifications.

Analog signals can be connected anywhere, there can be limitations imposed by the final application or the global architecture, but there is not any prohibition.

dorvan, 2021-06-28:
@M10CUBE AnthC it's not an HAT for Raspberry. AnthC it's a platform.

closed 2022-06-22 by clearemc